roberttidey/LightwaveRF #3: Raspberry Pi LwRx Test

dansinclair25, 2015-01-13:
Hi,

I was wondering whether you could upload the LwRX Test code for the Raspberry Pi? I've tried to get it working myself but to no avail!

Cheers,

/Dan

roberttidey, 2015-01-13:
These libraries are intended for minimal processor platforms like Arduinos and Arm Spark cores where there is good access to the hardware and interrupt systems. They weren't intended for Raspberry Pi with a full multi-tasking Linux type OS where it is more difficult to get guaranteed response time.

I have used Raspberry Pi with a receiver and some Python code to capture raw bit streams when doing protocol investigations on new devices so it is possible to do captures but it won't necessarily be as reliable as using lower level devices. I could make that available but it is intended to just capture the raw bit stream and doesn't do real protocol decoding. The Raspberry Pi pigpio library does give a method for doing fast captures so if I was going to try to support the TX and RX directly on the Pi, I think I'd be using that to do the raw capture and then putting my state machine based LW code on top of that.

ant-thomas, 2015-01-14:
I have used an Arduino connected to a Raspberry Pi over USB to control LWRF switches. Using the examples by @roberttidey on the Arduino then sending codes across the serial connection from the Raspberry Pi to the Arduino.

eg: `echo "4,1,159,1,1" > /dev/ttyUSB0`

dansinclair25, 2015-01-14:
Thanks for the feedback.

WHat I'm trying to do is read the signal from the LWRF Door Sensors - I know the Pi is "seeing" the signal (an LED hooked up to the VCC and DATA on the receiver shows this) but no library (except for piscope) can show me the Pi recognises the signal. I was hoping this library might do just that.

roberttidey, 2015-01-15:
As mentioned by others a lot of people use Arduinos to off load the low level bit encoding from the Raspberry Pi. In principle the Pi can do this directly, particularly if using the pigpio library which uses the DMA controller to read / write io in a real time fashion.

http://abyz.co.uk/rpi/pigpio/examples.html has examples of doing this. The Virtual Wire example is particularly interesting here as it is an interface to the general purpose 433MHz protocol used to communicate between Arduinos. This will not decode the LightwaveRF protocol but if the protocol part of the Virtual Wire was replaced by similar logic to the LightwaveRF library here then it should be possible to make that work.

roberttidey, 2015-01-17:
I have had a first go at doing direct Raspberry Pi support using the pigpio library. The RX side is working allowing a Raspberry with a simple 433MHx RX module to see LightwaveRF packets. I am still getting the TX side to work.

roberttidey, 2015-01-18:
TX side now working

hxpaul, 2015-01-19:
This is great, thanks a lot for this. I have successfully used your arduino libs and am now upgrading to the pi... I can't figure out from your code which gpio pins to connect the hardware too though, could you advise please? And how are you connecting, just with jumpers? Can you have both Rx and Tx components connected at once on the same pi? I have these two https://www.coolcomponents.co.uk/catalogsearch/result/?q=434mhz
If I have failed to read the manual please tell me!

roberttidey, 2015-01-19:
The library code has a main routine down the bottom which runs if the python file is just executed itself. In that routine when it initiates the RX and TX classes then it defines which pins to use. In this case it defines RX as GPIO24 and TX as GPIO25. 

So I have the receiver powered from 3.3V/GND and its data pin to GPIO24. For the TX I have it powered from +5V/GND and data pin to GPIO25. That is quite safe as the TX data is just an input and can't feed back excess volts to the Pi. I have them both connected at the same time and when the test is run, it transmits a number of packets and the receiver shows them being received. It will then display any more packets from other Lightwaverf controls it sees.

I am doing a bit more optimisation as in Python it is consuming quite a lot of cpu. If I can't get it significantly lower then I may do a C++ version of the Raspberry library.

dansinclair25, 2015-01-20:
Thanks to your python script I have my 2 LWRF Door Sensors being seen by my Pi perfectly (with a slight amendment to have it continually run).

FYI, my CPU usage is at 100% when running the script (but that's to be expected, right?!).

Thank you so much!

roberttidey, 2015-01-20:
I'm not happy with it using so much CPU. PIGPIO itself is pretty efficient but the python library interface to it is adding a lot of overhead. The RX receivers generate noise transitions all the time which cause continuous calls through the python handler. Even with a null protocol handler this noise is using 90% CPU. I am looking at various strategies to overcome this.

roberttidey, 2015-01-21:
Added c++ version of Raspberry code. This is way more efficient than the python version and CPU utilisation is now very low (<7%)

roberttidey, 2015-02-06:
A 3rd superior method for Raspberry Pi usage is added using the recent custom extension method to pigpio. This allows easy access from python whilst maintaining the low overhead of the c method.