roberttidey/LightwaveRF #4: Documention Question

markvr, 2015-02-07:
Hi, many thanks for your work on this!  I've been trying it out with a raspberry Pi and a Lightwave dimmable light switch.  It has paired OK and I can change the brightness (yay!), but not always predictably to what I want!

TL;DR:  In the python example, what should the values in: TX_TEST = [0,0,0,1,15,1,5,10,12,2], be to turn a light from "off" to "maximum brightness"?

I've read the PDF several times and am confused at the description on page 5 of the nibbles and commands.  My understanding is a "nibble" is 4 bits, so has a maximum value of 16, but you have parameters for "nibble 0" up to 256, which is obviously a full byte.  The obvious thing is too use "nibble 2" as well (e.g. a value of 159 (max brightness ) == 1001 1111 == [9, 15,...]), but nibble 2 has other params described, which also go up to 256.

Or am I barking up the wrong tree entirely with my calculations..?!

roberttidey, 2015-02-07:
If you check the first paragraph on page 4 just under the commands and parameters section it says the  first 2 nibbles are taken together as a parameter value (0-255). The headings on the table on page 5 could be misleading. The column labelled  as nibble actually is the  the value of the command nibble not the nibble index number as you may be reading it. So a command (4th nibble) has OFF value of 0, an ON value of 1, and a MOOD value of 2. I'll relabel that to make it clearer.

A typical 100% ON sequence as transmitted by wifi link for a dimmer is 9,14,d,1,a,a,a,a,a,r the first 2 nibbles give a value of 9*16+14 = 158 (like the 9th row in the table). I don't know why it doesn't use the maximum value of 159, but I don't think it makes much difference. The 4th nibble is a 1 (ON), d is the device number, r is the room number, and the a's are the base pairing address.

markvr, 2015-02-08:
Thanks for clarifying that.  I was indeed reading the "nibble" column on page 5 as the index of the nibble in the array, and hence confused about how the first two nibbles (which combine into one byte value as described on page 4) corresponded to the table on page 5. 

It now works perfectly, thanks :)

closed 2015-02-08 by markvr