roberttidey/LightwaveRF #12: How to pass parameters in command message?

LondonSi72, 2018-01-05:
Hi,

Sorry, wasn't sure where else to ask this question.  I have a Lightwaverf dimmer that I'm trying to control through a Raspberrypi using your libraries - I'm very much a newbie at this, so not sure how to form the command sequence to tell the dimmer to dim, not just turn on or off.  What should the entry in tx.put be to set a dimmer level?

Many thanks!

roberttidey, 2018-01-05:
In the PDF document in a section called commands and parameters it describes the structure of a message including parameters. The table in there shows that to set a brightness level you want to send an ON command with a parameter value 128 and 159. Other ranges starting at 64, 96 etc have the same effect but starting from 128 is what Lightwaverf controllers tend to use. So these parameter gives 32 possible values giving levels from 0 to 100% in about 3% steps.

To fully form the message to send you need add the room and device number and the unique Identity.

There are 2 strategies you can use here for that depending on whether you have a Lightwaverf link or whether you are running in standalone mode.

For standalone mode you can put whatever you like in the room, device and id fields and then pair the device (dimmer switch to this). Put the device in pairing mode and then initiate a send from the raspberry Pi. The device will remember the room, device and id fields, and respond to future messages with these fields but with different commands and parameters. For further devices keep the id fields the same and change the room and device fields that you pair them to. You are effectively building your own link by doing this. Note that devices can remember up to 6 different pairings so it is not a problem if you are already using remote controllers or a link.

If you have a link already and use the phone app then it is advantageous to get the raspberry Pi to use the same ids and room and device numbers the Link already uses. The raspberry Pi is then effectively issuing exactly the same type of commands the link would. For devices already paired to the link no further pairing is needed as the devices can't tell whether the message came from. The easiest way to set this up is to have a 433MHz receiver on the Pi and use the Rx library to display what it gets when the link sends a message. That way you can just copy the ids and room and device numbers. The lwrf.py test program prints out any messages it receives for 30 seconds (after going through the tx test) so that can be used to see these codes.

As an example say you want to set a device to 40%. 40% is about 13 in the 0-32 range. So a message would look like

8,13,0,1,4,5,6,7,8,2

where the 8,13 is the paramter 8*16 + 13 or 13 units in the dim range
0 is the device number
1 is the ON command
4,5,6,7,8 are the 5 id numbers
2 is the room number

LondonSi72, 2018-01-06:
Thanks Robert for the really fast reply. I'll try that later today. 

I had read your pdf but I couldn't work out how to add the parameters to the command message. Your example really helps. 

(I don't have a link - just the dimmer. I'm trying this to see if it's worth investing in more devices to add to my home automation. I have a Google AIY kit from MagPi and have successfully set up custom Python commands for TP link and Edimax switches. I want something to manage a room with multiple ceiling lights now without shelling out for loads of smart bulbs)

I'll let you know how I get on.

LondonSi72, 2018-01-06:
Hi again.

Thank you - that worked perfectly.  (Somehow I had  missed the fact the first 2 nibbles are the parameters even though you state it quite clearly in the pdf!)

I also, coincidentally, received a remote control that I bought from eBay today, so I could read the messages sent from it.  It sends BF01 as the increase brightness parameter, and A000 as the dimming parameter.  Using your suggested parameters works great for setting an absolute brightness, and is what I would use practically rather than an increase or decrease brightness command.

Lastly - and I don't know if this is the age of the remote control, the quality of the RX I'm using, or jitter in the Pi (it's a model 3 running Jessie) - but there was quite a lot of variation in the messages displayed as received by the Pi.  In particular the id varied quite a lot - is that something you have seen?  (I am using your first method of implementation/ library as that was simplest to test - so not the one that involves making pigpio with the custom extension.)

Thanks again.

roberttidey, 2018-01-06:
Good to hear you are having some success.

RX modules do vary quite a bit in quality and that can have some impact on the reliability of messages received. The very cheap ones which have a little green inductor in the middle use a super-regenerative receiver. This is sort of OK but the ones that say they are a superhet or superheterodyne design are much better and still pretty cheap. It is important to have antennas on the TX and RX if you want to get good range. Either a straight wire of about 17cm or same formed into a coil work well.

BF01 and A000 sound right and correspond to increase / decrease brightness in the PDF table.

The first method of raspberry lightwaverf programming is very processor intensive. It works OK particularly on later Raspberry models but is susceptible to an occasional glitch because the linux OS can sometimes task switch in the middle of a message.

The pigpio methods do not suffer from this and use much less cpu power. I know they look more complicated but they are not too bad if you follow the instructions.

LondonSi72, 2018-01-07:
Thanks for the further info.  Yes my receiver looks to be one of the very cheap ones.  I'm not really using the receive side for my projects, so probably not going to be a problem.

I'm still not 100% clear on generating the right paramaters, but through trial and error I finally managed to get a set of parameters that seems to adjust the brightness to roughly 10, 25, 33, 50 and 75% plus a max and a min and a brighten and a dim.  Took a bit of play around as trying to get a brightness of _30% or more required a 'nibble' that was greater than 15, and putting, for example, 8,15 in the transmission message ended up with a message that was translated to 8,5 when sent, so I had to play around with the first nibble too.

For anyone who's interested and reads this, here are the first two nibbles paramaters I used for varying brightness levels (in decimal):

min [6,0
10% [8,3
25% [8,8
33% [8,13
50% [9,7
75% [9,13
max [7,15

dim [10,0
brighten [11,15

The latter need to be transmitted over 100 times to dim or brighten across the whole range (I even found the remote control won't fully brighten or dim a bulb through the whole range)

I'll close this 'issue' now.  Many thanks for your help - much appreciated :)

Simon

closed 2018-01-07 by LondonSi72