roberttidey/LightwaveRF #13: LwTxTest has multiple compile errors with Arduino 1.8.5

ggodart, 2018-03-04:
I tried compiling LwTxTest and get following errors, am I missing an include?;

Arduino: 1.8.5 (Mac OS X), Board: "Arduino Nano, ATmega328P"

.../libraries/LwTx/LwTx.cpp: In function 'void lw_timer_Setup(void (*)(), int)':
.../libraries/LwTx/LwTx.cpp:284:18: error: 'timer1_isr_init' was not declared in this scope
  timer1_isr_init();
                  ^
.../libraries/LwTx/LwTx.cpp: In function 'void lw_timer_Start()':
.../libraries/LwTx/LwTx.cpp:294:35: error: 'timer1_attachInterrupt' was not declared in this scope
  timer1_attachInterrupt(isrRoutine);
                                   ^
.../libraries/LwTx/LwTx.cpp:295:16: error: 'TIM_DIV16' was not declared in this scope
  timer1_enable(TIM_DIV16, TIM_EDGE, TIM_LOOP);
                ^
.../libraries/LwTx/LwTx.cpp:295:27: error: 'TIM_EDGE' was not declared in this scope
  timer1_enable(TIM_DIV16, TIM_EDGE, TIM_LOOP);
                           ^
.../libraries/LwTx/LwTx.cpp:295:37: error: 'TIM_LOOP' was not declared in this scope
  timer1_enable(TIM_DIV16, TIM_EDGE, TIM_LOOP);
                                     ^
.../libraries/LwTx/LwTx.cpp:295:45: error: 'timer1_enable' was not declared in this scope
  timer1_enable(TIM_DIV16, TIM_EDGE, TIM_LOOP);
                                             ^
.../LwTx/LwTx.cpp:296:24: error: 'timer1_write' was not declared in this scope
  timer1_write(espPeriod);
                        ^
.../libraries/LwTx/LwTx.cpp: In function 'void lw_timer_Stop()':
.../libraries/LwTx/LwTx.cpp:308:17: error: 'timer1_disable' was not declared in this scope
  timer1_disable();
                 ^
.../libraries/LwTx/LwTx.cpp:309:25: error: 'timer1_detachInterrupt' was not declared in this scope
  timer1_detachInterrupt();
                         ^
exit status 1
Error compiling for board Arduino Nano.

roberttidey, 2018-03-04:
The library can be used in multiple environments (Spark Core, Due, Pro32u4, AVR328, ESP8266)
Nano is the same as AVR328

The LwTx.h file has defines at the top to decide which to use.
I think you have got it to be defined for ESP8266 which is the default. This makes it look for ESP8266 timer routines which won't exist in the nano environment.

For Nano you need to use the AVR328 version. So you need to set the defines like
//#define SPARK_CORE 1
//#define DUE 1
//#define PRO32U4 1
//#define AVR328 1
//#define ESP8266 1
#define AVR328 1

Make sure that AVR328 is the only one in that group defined.

ggodart, 2018-03-05:
Thanks Robert for the quick response
Now have clean compile
Giles

closed 2018-03-05 by ggodart